YosysHQ/icestorm #255: make install fails with BSD sed on macOS

mathisschmieder, 2020-05-26:
Hi,

While trying to install icestorm on macOS, I ran into some incompatibilities between GNU and BSD sed. The changes in https://github.com/cliffordwolf/icestorm/commit/fe3086a7339f0c2731fb1a14f6ef0e4d93a0377e break the make install process on macOS:

`sed -i 's+import iceboxdb+import iceboxdb as iceboxdb+g' icebox.py
sed: 1: "icebox.py": command i expects \ followed by text`

The fix is actually quite easy, you just have to add -e after -i:

`sed -i -e 's+import iceboxdb+import iceboxdb as iceboxdb+g' icebox.py`

This will work both on BSD and GNU sed (tested on macOS Catalina 10.15.4 with BSD sed and GNU sed version 4.8 and Ubuntu 20.04 with GNU sed version 4.7). 

Thanks a lot,
Mathis

mathisschmieder, 2020-05-26:
There was an error in my proposed fix, this will create a backup file with extension '-e'. The proper fix is as follows:

`sed -i'' -e 's+import iceboxdb+import iceboxdb as iceboxdb+g' icebox.py`

parasyte, 2020-06-02:
I hit the same problem last night and used the latter suggestion in my local build.

derekmulcahy, 2020-06-08:
Worked for me as well, thanks.

mmicko, 2021-09-06:
Addressed in https://github.com/YosysHQ/icestorm/commit/9fe28369aaf1f98090b9ecfded6ef22599cc09d0

closed 2021-09-06 by mmicko