YosysHQ/icestorm #258: Error: Got .io_tile statement for logic tile

zzdts, 2020-06-20:
At the end of make , icepack errors happen:
```
icepack top.asc top.bin
Error: Got .io_tile statement for logic tile 1 -1855121696.
```

platform: macos Catalina 10.15.5

daveshah1, 2020-06-20:
This looks similar to #231, but I don't know if the root cause was ever identified - do you remember if you fixed it @xobs ?

xobs, 2020-06-20:
Never did figure out the root cause of it. Those packages were attempts to build this project under Windows and not cross-compiled so we could build it under Conda, but there were other issues with that build system so I abandoned the attempt and never looked into what caused that problem.

zzdts, 2020-06-21:
Asc file  line ending with unix type.   
I cp top.asc to ubuntu and run "icepack top.asc top.bin" ,it works well.

zzdts, 2020-06-22:
I fond this project and the build system support macos  [https://github.com/FPGAwars/toolchain-icestorm](https://github.com/FPGAwars/toolchain-icestorm)
It's works well on Mac.
@xobs  @daveshah1

closed 2020-06-22 by zzdts

xobs, 2020-06-22:
We've taken that and adapted it to work with more tools, including nextpnr. The resulting toolchains are built daily and are available at https://github.com/open-tool-forge/fpga-toolchain