YosysHQ/icestorm #268: enhancement request: Support the NVCM

sxpert, 2020-08-18:
title says it all

smunaut, 2020-08-24:
If you could record the SPI communication when programming the NVCM of some part using the lattice tools, that would be very helpful.

Unfortunately I don't have a bunch of parts to waste programming the NVCM ...

majbthrd, 2020-10-06:
@sxpert , I'm just added Pull Request #272 with a tool to generate .nvcm files from iCE40 .bin files.  I've used its output with Lattice Diamond Programmer, for what that is worth.  I believe it to be functional and have tested it against existing .bin and .nvcm files, but perhaps you could put a second set of eyes on it?

I previously added a Pull Request #271 that reads a special part of the NVCM that contains a device ID, and I have test code locally that reads the entire contents of the NVCM (if not secured).

I believe that I know exactly what the steps are to do the programming (the bulk of the steps are prescribed in the .nvcm itself), but as @smunaut said, one could go through a lot of parts in testing.

ryan1109, 2022-06-26:
I have a question，Why is it 7eAA997e010e?  Instead of "70807e99557e"。

ryan1109, 2022-06-26:
> #286 is an initial go at a tool to program the NVCM on the iCE40 chips. I've only bricked two so far in developing it...

I have a question，Why is it 7eAA997e010e? Instead of "70807e99557e"。