YosysHQ/icestorm #269: Targetting 8k-tq144:4k but missing pins 65 and 66 which are bound to ICE_CDONE and ICE_CREST.

46cv8, 2020-09-28:
I suspect this is probably just my lack of experience using icestorm and not an actual issue, but I'm not sure where else would be appropriate to ask.

I just received the project files from a friend who is using icecube2 to build a project and in his pcf file he references pins 65 and 66 as follows and is targetting the package 8k-tq144:4k.
`set_io ICE_CDONE 65
set_io ICE_CREST 66`

These pins appear to be missing from icebox.py at the corresponding location for 8k-tq144:4k.
https://github.com/YosysHQ/icestorm/blob/da52117ccd5b4147f64dc7345357ec5439cd7543/icebox/icebox.py#L3704

I don't have a license for icecube2 so I can't perform the steps suggested in the doc string.
`# This dictionary maps package variants to a table of pin names and their
# corresponding grid location (x, y, block). This is most easily found through
# the package view in iCEcube2 by hovering the mouse over each pin.`

I can confirm however that there appear to be some only two pins missing as I count 107 pins defined in icebox.py, but the documentation below says there are a total of 109 pins with 2 dedicated io pins.
http://www.latticesemi.com/view_document?document_id=9

To compare I also took a quick look at 1k-tq144 in icebox.py and it shows 96 pins defined. Meanwhile in the ProductSelectorGuide it shows 98 pins and 2 dedicated pins.

I tried just commenting out the two definitions that were using pins 65 and 66 to see if I could get things working.

But I then get an error from nextpnr_ice40 as follows.
`ERROR: IO 'ICE_CDONE' is unconstrained in PCF (override this error with --pcf-allow-unconstrained)`

It seems like I'm missing something and that perhaps there is some other name I should be using to bind these two pins rather than the pin numbers 65 and 66 that was used in the project I received that was building with icecube2. Or perhaps I am supposed to use --pcf-allow-unconstrained. I suspect this is because they are dedicated IO's perhaps and thus excluded from the io's that one can manually bind too.

Any help would be appreciated.

daveshah1, 2020-09-28:
icestorm doesn't have a way of representing these pins as they don't have an IO tile associated with them (and unless I'm missing something, can't be used in the design for anything). The best solution is just to remove them from the PCF and design.

46cv8, 2020-09-28:
Thanks for the super fast response and solution, after looking at the design it didn't appear the pins were used for anything so I just removed them and it built fine!
Closing out this issue.

closed 2020-09-28 by 46cv8