YosysHQ/icestorm #273: Not able to use both PLLs of iCE40-HX4K

esynr3z, 2020-11-22:
My goal is to create 4 different clock domains from one 12 MHz input clock:

- 12 MHz
- 40 MHz
- 128 MHz
- 64 MHz

Chip is iCE40-HX4K in TQ144 package.

Tried using different banks.
pinout.pcf:
```verilog
# bank 3 clock
#set_io gen_12mhz 21

# bank 0 clock
set_io gen_12mhz 128

set_io pll1_lock 1
set_io pll2_lock 3
```

Tried using differnt PLL configurations.
top.v:
```verilog
module top(
    input  wire gen_12mhz,
    output reg  pll1_lock,
    output reg  pll2_lock
);

wire clk_12mhz;
wire clk_128mhz;
wire clk_64mhz;
wire clk_40mhz;

SB_PLL40_2_PAD #(
    .FEEDBACK_PATH("SIMPLE"),
    .DIVR(4'b0000),
    .DIVF(7'b0110100),
    .DIVQ(3'b100),
    .FILTER_RANGE(3'b001)
) pll1 (
    .LOCK          (pll1_lock),
    .RESETB        (1'b1),
    .BYPASS        (1'b0),
    .PACKAGEPIN    (gen_12mhz),
    .PLLOUTGLOBALA (clk_12mhz),
    .PLLOUTGLOBALB (clk_40mhz) // 39.750 MHz actually
);

SB_PLL40_2F_CORE #(
    .DIVR (4'b0000),
    .DIVF (7'b1010100),
    .DIVQ (3'b011),
    .FILTER_RANGE (3'b001),
    .FEEDBACK_PATH ("SIMPLE"),
    .PLLOUT_SELECT_PORTA ("GENCLK"),
    .PLLOUT_SELECT_PORTB ("GENCLK_HALF")
) pll2 (
    .LOCK           (pll2_lock),
    .RESETB         (1'b1),
    .REFERENCECLK   (clk_12mhz),
    .PLLOUTGLOBALA  (clk_128mhz), // 127.5 MHz actually
    .PLLOUTGLOBALB  (clk_64mhz)   // 63.75 MHz actually
);

endmodule
```

Synthesis is ok. But nextpnr-ice40 gives an error:
```
$ nextpnr-ice40 --seed 42 -l pnr.rpt --pre-pack sdc.py --pcf pinout.pcf --hx8k --package tq144:4k --json top.json --asc top.asc
Info: constrained 'gen_12mhz' to bel 'X17/Y33/io0'
Info: constrained 'pll1_lock' to bel 'X0/Y30/io1'
Info: constrained 'pll2_lock' to bel 'X0/Y28/io1'
Info: constraining clock net 'clk_12mhz' to 12.00 MHz
Info: constraining clock net 'clk_128mhz' to 128.00 MHz
Info: constraining clock net 'clk_64mhz' to 64.00 MHz
Info: constraining clock net 'clk_40mhz' to 40.00 MHz

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info:        0 LCs used as LUT4 only
Info:        0 LCs used as LUT4 and DFF
Info: Packing non-LUT FFs..
Info:        0 LCs used as DFF only
Info: Packing carries..
Info:        0 LCs used as CARRY only
Info: Packing RAMs..
Info: Placing PLLs..
ERROR: PLL 'pll1' PACKAGEPIN SB_IO 'gen_12mhz$sb_io' is not connected to any PLL BEL
ERROR: Packing design failed.
0 warnings, 2 errors

```

Is there any way to get over it?

All sources and scripts:
[ice40hkx_2_pll_issue.zip](https://github.com/YosysHQ/icestorm/files/5579387/ice40hkx_2_pll_issue.zip)

daveshah1, 2020-11-22:
Pin 128 is not a dedicated PLL input, if you change that to 129 then it works.

esynr3z, 2020-11-22:
Wow, thanks! I tried to find this information with no success.

Just to be sure. If I use non-dedicated PLL input for clock, I'm able to use the only one CORE PLL. Right?

daveshah1, 2020-11-22:
You can use two CORE PLLs if you want to. But beware that whenever you use a CORE PLL, the corresponding dedicated PLL input pin becomes reserved and cannot be used as a general purpose input (it can still be used as a general purpose output though).

esynr3z, 2020-11-22:
I think I faced exactly the problem you described. What actually inputs pins are dedicated to PLL?
129 and 49?

I use SB_PLL40_CORE + SB_PLL40_2F_CORE in my project, and nextpnr gives me an error:

```verilog
Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info:     1128 LCs used as LUT4 only
Info:      507 LCs used as LUT4 and DFF
Info: Packing non-LUT FFs..
Info:      312 LCs used as DFF only
Info: Packing carries..
Info:      418 LCs used as CARRY only
Info: Packing RAMs..
Info: Placing PLLs..
Info:   constrained PLL 'pll1' to X16/Y33/pll_3
ERROR: PLL 'pll2' couldn't be placed anywhere, no suitable BEL found.
ERROR: Packing design failed.
4 warnings, 2 errors
Can't open input file: No such file or directory
Error: Failed to open input file.
```

In this project almost all IOs are in use, so I can guess that I'm trying to use as in input some PLL related pin.

esynr3z, 2020-11-23:
## TL;DR

Yo have to pay more attention to Bank 0 and Bank 2 GBIN pins, especially if you want to use PLLs AND use free clock buffer pins as IOs.

## Detailed description

If you want to use ice40_PLL_PAD, ice40_PLL_2F_PAD or ice40_PLL_2_PAD you have to route external clock only to:

* IOB_81_GBIN5 - pin 49
* IOT_198_GBIN0 - pin 129

I didn't found that information in any vendor PDF's or pinout files, just figured out it experimentally.

If you want to use ice40_PLL_CORE or ice40_PLL_2F_CORE you won't be able to use PLL dedicated pins as inputs.

Some additional information from iCE40 sysCLOCK PLL Design and Usage Guide (FPGA-TN-02052-1.2):

```
5.1.2. iCEcube2 Software

* If any instance of PLL is placed in the location of the I/O cell, then, an instance of SB_GB_IO cannot be placed in
that particular I/O cell.

* If an instance of ice40_PLL_CORE or ice40_PLL_2F_CORE is placed, an instance of SB_IO in output-only mode can
be placed in the associated I/O cell location.

* If an instance of ice40_PLL_PAD, ice40_PLL_2F_PAD, ice40_PLL_2_PAD is placed, the associated I/O cell cannot be
used by any SB_IO or SB_GB_IO.

* If an instance of ice40_PLL_2F_CORE, ice40_PLL_2F_PAD, ice40_PLL_2_PAD is placed, an instance of SB_IO in
output-only mode can be placed in the right neighboring I/O cell.
```

closed 2020-11-23 by esynr3z