YosysHQ/icestorm #281: iceprog takes 165secs to program 111kB into UPduino with FT232H

igor-m, 2021-03-10:
I do not know whether there is a problem with my hw (UPduino 2.0 with FT232H) or with iceprog itself, but flashing 104+7kB of binary takes 165 seconds here (Ubuntu 18.04 in VirtualBox_Win10):

```
me@UBU:~/picorv32/picosoc$ make icebprog
iceprog icebreaker.bin
init..
cdone: high
reset..
cdone: high
flash ID: 0xEF 0x40 0x16 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
programming..
reading..
VERIFY OK
cdone: high
Bye.
iceprog -o 1M icebreaker_fw.bin
init..
cdone: high
reset..
cdone: high
flash ID: 0xEF 0x40 0x16 0x00
file size: 6708
erase 64kB sector at 0x100000..
programming..
reading..
VERIFY OK
cdone: high
Bye.
me@UBU:~/picorv32/picosoc$ 

```
Is that ok?

gatecat, 2021-03-10:
iceprog is very slow with VirtualBox in USB 1.1 mode for some reason. Try with the extension pack and USB 2.0/EHCI mode.

igor-m, 2021-03-10:
Oracle VM_Virtualbox latest (in Win10), Ubuntu 18.04:
```

VM_USB_setting    iceprog 111kB time secs
USB 1.1                 165
USB 2.0                 165
USB 3.0                 15
```
Tried on all USB ports here (hp probook 450)..

mmicko, 2021-09-06:
As confirmed this is VM usage related issue.

closed 2021-09-06 by mmicko