YosysHQ/icestorm #292: Errors when running post-synthesis simulation examples

strawberryhacker, 2022-03-23:
Hi, I want to do post-synthesis simulation for the icebreaker. I tried running `make rs232demo_syntb.vcd`, but I get errors when processing `cells_sim.v`. The full command that results in the errors below is `iverilog -o rs232demo_syntb rs232demo_tb.v rs232demo_syn.v 'yosys-config --datdir/ice40/cells_sim.v'`

This is the output: 

```
/usr/local/share/yosys/ice40/cells_sim.v:20: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:127: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:179: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:305: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:592: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:650: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:727: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:785: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:902: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1189: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1247: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1324: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1382: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1462: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1701: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1837: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
/usr/local/share/yosys/ice40/cells_sim.v:1974: syntax error
/usr/local/share/yosys/ice40/cells_sim.v:1: Errors in port declarations.
```

It seems the problem is due to the DEFAULT_ASSIGNMENT defines. The first error from above comes from the third input, see below.

```
module SB_IO (
	inout  PACKAGE_PIN,
	input  LATCH_INPUT_VALUE,
	input  CLOCK_ENABLE `ICE40_DEFAULT_ASSIGNMENT_1,
	input  INPUT_CLK,
	input  OUTPUT_CLK,
	input  OUTPUT_ENABLE,
	input  D_OUT_0,
	input  D_OUT_1,
	output D_IN_0,
	output D_IN_1
);
```

where

```
`define ICE40_DEFAULT_ASSIGNMENT_1 = 1'b1
```

Does anyone know how to make this work? I would like to be able to simulate with various build in modules such as `SB_PLL40_PAD`.

maxekman, 2022-10-25:
I have the same issue, any pointers would be welcome!

mmicko, 2022-10-26:
iverilog does not fully support standard, so there is need for you  to add `-DNO_ICE40_DEFAULT_ASSIGNMENTS` to command line for iverilog as in https://github.com/YosysHQ/yosys/blob/master/tests/arch/run-test.sh#L14 to make this work

closed 2022-10-26 by mmicko