YosysHQ/icestorm #300: Add support for iCE65 devices?

nonarkitten, 2022-08-19:
It's rumoured these devices still exist even though Lattice seems to have gone out of their way to bury them. None of the development tools are available anymore, not from any site, so unless you were lucky to grab a copy then, you're out of luck. Supposedly, these are little different from the first iCE40 chips, just in a larger process that's 5V tolerant.

gatecat, 2022-08-19:
Not sure if any of the icecube versions in Lattice's archive here might support it - they go back to 2012
https://www.latticesemi.com/Support/SoftwareArchive (check the downloadable software tab)

The chips aren't too hard to find at good old Rochester: https://www.rocelec.com/part/SLBICE65L08F-TCC72I

matthiasbock, 2022-08-19:
Also, products with iCE65 FPGA are still being sold. Most relevant for me personally would be the SeeedStudio / SainSmart Handheld Oscilloscope DSO203, which features a STM32F103 alongside an iCE65L04. PCB schematic as well as source code for both MCU and FPGA firmware are available. Mods for the MCU firmware exist also, but to my knowledge none for the FPGA.

* https://www.sainsmart.com/products/dso203-4-ch-handheld-mini-digital-oscilloscope
* https://wiki.seeedstudio.com/DSO_Quad/
* https://github.com/Seeed-Studio/DSOQuad_SourceCode
  * https://github.com/matthiasbock/dso203

According to Lattice support, iCE65 FPGAs were discontinued in 2012 and the latest software supporting them is iCECube2_2011_12. They provide a temporary download link upon request.

nonarkitten, 2022-08-19:
I thought 2012.03 was the last one that supported the iCE65, but yeah, the archive does not have a version going far enough back.

nonarkitten, 2022-11-15:
After bugging Lattice for a month I was basically able to get nothing. I have the old version of iCEcube2 (c. 2011) but without a license file, it's not very useful. So if there's any hope for the iCE65's, it's here.

matthiasbock, 2022-11-15:
It should be possible to request a license here:

* https://www.latticesemi.com/Support/Licensing

To my knowledge Lattice provides them free of charge.

nonarkitten, 2022-11-15:
Yes, thank you, I am WELL aware of their free license.

The file you get from them will not work on the old version of iCEcube2, it reports an error in the license file and will not work.

Lattice will then blame Synopsis for this since it's their backend. Synopsis came back and blamed Lattice for changing the feature labels. And then nothing gets fixed. This has been a dead support issue now for over a month.

nonarkitten, 2025-11-26:
This aged like milk -- Lattice has removed their Free License entirely. I was able to get the license but there are tens of thousands of iCE65's sitting in cold storage at Rochester we could be using all with bona fide 5V tolerance.