YosysHQ/icestorm #301: yosys regression, cause icebram to fail

bjoto, 2022-09-09:
### Version

Yosys 0.21+1 (git sha1 b7bf68501, gcc 11.2.0-19ubuntu1 -fPIC -Os)

### Reproduction Steps

I'm getting some strange behavior with yosys >0.17+16, and I think d7dc2313b915 ("ice40: Use `memory_libmap` pass.") might be the issue.

To reproduce, clone icestorm (https://github.com/YosysHQ/icestorm.git), and build `icebram`.
```
cd icebram
make
while true; do bash ./rundemo.sh |& grep 'Found and replaced '; done
```

### Expected Behavior

`Found and replaced` string should never be zero. All yosys <= 0.17+16 outputs e.g.:

```
Found and replaced 1 instances of the memory.
Found and replaced 1 instances of the memory.
Found and replaced 2 instances of the memory.
Found and replaced 3 instances of the memory.
Found and replaced 3 instances of the memory.
Found and replaced 2 instances of the memory.
Found and replaced 2 instances of the memory.
Found and replaced 1 instances of the memory.
Found and replaced 1 instances of the memory.
Found and replaced 1 instances of the memory.
Found and replaced 3 instances of the memory.
```

### Actual Behavior

icebram should be able to find a "ram_data" section to replace, but almost always, it fails to do so:
```
$ while true; do bash ./rundemo.sh |& grep 'Found and replaced '; done
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 1 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 4 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 3 instances of the memory.
Found and replaced 0 instances of the memory.
Found and replaced 0 instances of the memory.
^C
```

nakengelhardt, 2022-09-09:
I've transferred the issue; I'll see if I can find someone more familiar with icebram to take a look.

lethalbit, 2022-09-14:
@bjoto The current example uses the deprecated `arachne-pnr`

I've updated it on my branch to use `nextpnr-ice40` and i've not run into this problem, using the latest Yosys and nextpnr.

If you replace the old `rundemo.sh` with this one do you still get that error?

```bash
#!/bin/bash
set -ex

python3 makedemo.py
yosys -p 'synth_ice40 -json demo.json' demo.v
nextpnr-ice40 --hx8k --package bg121 --pcf-allow-unconstrained --asc demo.asc --json demo.json

./icebram -v demo_dat0.hex demo_dat1.hex < demo.asc > demo_new.asc
```
The PnR can't always place with `--pcf-allow-unconstrained` but as the old exampled used a  `demo.pcf` that was not committed or generated this is needed to get an output bitstream asc.

So far on my runs i've not run into an issue where `icebram` can't find any memories to replace using this updated example. Do let me know if it is still happening.

lethalbit, 2022-09-14:
So far I've found that it looks like it is memory geometry dependent, still working on narrowing it down. but for instance, the example with the following geometry will fail:
* bram_width: 12
* bram_depth: 1024
* num_prorts: 1

while the following geometry will always succeed:
* bram_width: 12
* bram_depth: 1280
* num_prorts: 2

The reason the examples seem to fail randomly is because of how the demo is generated, by picking a random width, depth, and port count within a range.

Currently looking into see if this is a synthesis issue or an icebram issue

lethalbit, 2022-09-14:
Yeah I suspect it's an icebram issue, looking into seeing what could possibly cause it at the moment.

niels-moller, 2022-11-20:
Thanks for filing this issue. I suspect it's the reason icebram doesn't work for me. I'm using Yosys 0.22+57 (git sha1 518194fac, gcc 10.2.1-6 -Og -fPIC), and my design includes a RAM of 1024 words, 64-bits wide.

As far as I understand, this is synthesized as 4x4 SB_RAM40_4K, each configured as 256 words of 16 bits. So both address space and data width are split in four pieces, and there are many ways that could be done (e.g., words could be split in 16-bit chunks, or interleaved, any two bits of the address could be used as a bank select).

I wonder if there's any possibility to have yosys propagate some metadata from the verilog (in my case) source file to the .ram_data lines in the generated .asc file, with an id to the register array in the sources, and how bits are mapped from the source array to this BRAM block?  Matching the random initialization data appears a bit brittle.

tomverbeure, 2022-12-27:
Eric just had a pretty long Yosys bisecting exercise to figure out why his existing design failed after I tried to recompile his example code. (See https://github.com/ICE-V-Wireless/ICE-V-Wireless/commit/38e0798b7f3acd68918e2c34cd632a507d861276.)

If it won't be fixed anytime soon, would it make sense to make icebram print a warning of some sort that there's a known issue?

niels-moller, 2022-12-27:
> If it won't be fixed anytime soon, would it make sense to make icebram print a warning of some sort that there's a known issue?

To me, it would make sense for icebram to treat "Found and replaced 0 instances of the memory." as a fatal error .

tomverbeure, 2022-12-27:
CAD flows (commercial or not) have a way to dumping thousands of warnings that are routinely ignored. 

“Treating” one be of those warnings as a fatal error is IMO not sufficient because it doesn’t halt the build of existing projects.

What about changing icebram so that it returns an error code so that most existing Makefiles will abort?

smunaut, 2023-02-14:
I rewrote icebram : See this branch in my tree https://github.com/smunaut/icestorm/tree/icebram
Testing would be appreciated to see if there are cases where it fails.

emeb, 2023-02-14:
This has worked for my failing case.

dirkenstein, 2023-02-16:
Works for me, too. At least, does not produce errors. Not tested the actual code yet.

chrbayer, 2023-02-17:
For me it works, too. No errors are produced and the design is working, too. So thank you very much!

PR #309 (2023-02-17): icebram: Complete rewrite to cleanup and make it work
Fixes #301 

I know it's a large diff, but it's pretty much a rewrite since the old structure of one giant main() function was becoming unwieldy.

icebram used to parse each .ram_data line one at a time but now it needs to load the whole memory. It also needs to know the READ_MODE (and so parse the ram config tile).
closed 2023-02-17 by gatecat