YosysHQ/icestorm #304: 'initial begin' bit order vs assigning value on clock cycle

chrisruk, 2022-11-12:
Hi,

I'm wondering if anyone might be able to help with a rather confusing issue I have, using yosys and an icestick FPGA board.

If I assign values to my BRAM within a clock cycle as in - https://github.com/chrisruk/matrixchip/blob/main/src/matrix.v#L54

The output bitstream, which goes to an 8x8 SK9822 LED matrix to display a font, but the letter appears upside down.

But if I move the font assignments to an initial begin statement, it appears the correct way up -

```
    initial begin
        fonts[0] = 64'he0_60_6c_76_66_66_e6_00;  // h
        fonts[1] = 64'h00_00_78_cc_fc_c0_78_00;  // e
        fonts[2] = 64'h70_30_30_30_30_30_78_00;  // l
        fonts[3] = 64'h00_00_78_cc_cc_cc_78_00;  // o
    end
```

Just wondering if anyone has any idea how this could happen?

In a simulation with each version, both output bitstreams appear the same.

msinger, 2023-04-24:
You can't access multiple memory cells in a BRAM within the same clock cycle. You need one clock cycle per access. Think about it, how should this work? Your data bus is 64 bit. How could this data bus carry four 64 bit values at the same time?

You can use the `(* mem2reg *)` attribute for your array definition. Then it is implemented with DFFs, not BRAMs. Then it works in parallel.

Initial block is different. The iCE40 allows the bitstream to contain an initial value for all the BRAMs. This value is written into the BRAM while the FPGA loads its bitstream via SPI during configuration, not during runtime. This is very hardware specific though. Not all FPGAs allow BRAMs to have initial values.

chrisruk, 2023-04-29:
Thanks a lot for your reply, that makes a lot of sense re. a single clock cycle.

Will close this now.

closed 2023-04-29 by chrisruk