YosysHQ/icestorm #315: Hacked Website

7unn3l, 2023-04-27:
I've noticed https://clifford.at/icestorm contains online gambling ads and generally seems pretty hacked. I couldn't find a contact for the site so I opened this issue, just to raise awareness. This is still the number one google result when I search for project icestorm

nakengelhardt, 2023-04-27:
We don't own the domain any more, it was registered by someone else who copied the old content to take advantage of existing search rank, there isn't anything we can do about that. The links we control should be changed already, but if you find a link to the old domain somewhere in the code, please point to the location so that we can update!

I believe it's also possible to report spam to google, perhaps if enough people do that it would reduce the search rank?

scherrey, 2023-04-27:
If someone could claim the copied website is violating their copyright or
trade identifier a DCMA take down request could be made against the hosting
provider. Who wrote the original content?

  - - Ben Scherrey

ErikSwan, 2024-04-19:
@clairexen, what is the status of your website? As the previous commenter pointed out, the current link for "more information" in the IceStorm README goes to a 404 page.