YosysHQ/icestorm #344: blinky example modified for "lp384" and "qn32" results in iverilog error: 'io_2_9_0' has already been declared in this scope

ericfont, 2025-02-26:
blinky.sh runs fine when run on a fresh git checkout of master for "hx1k" & "tq144".  However, the chip I have is 32-Pin QFN LP384, and so I tried to modify by changing those strings to "lp384" and "qn32" as (like https://github.com/YosysHQ/nextpnr/commit/e5b44026057f86b13499ec23e083e5a1c01ee4c3) and changed the pins (like [Image](https://github.com/user-attachments/assets/b709eb5c-9cfe-4eb5-a07d-a3dcd7d6d3bc)). I must be doing something simple wrong, but runing blinky.sh now results in an error at the last step when running iverilog:

`blinky_chip.v:27: error: 'io_2_9_0' has already been declared in this scope.`

(see log: [blinky_terminal_output.txt](https://github.com/user-attachments/files/18981089/blinky_terminal_output.txt) and [blinky_chip.v.txt](https://github.com/user-attachments/files/18981112/blinky_chip.v.txt) which has on line 3:

`module chip (input io_0_4_1, output io_2_9_0, output io_2_9_1, output io_5_9_0, output io_7_6_1, output io_6_9_0);`

and on line 27: 

`reg io_2_9_0 = 0;`

Probably something very simple I'm doing wrong. This error looks similar to https://github.com/YosysHQ/icestorm/issues/328

KrystalDelusion, 2025-02-26:
iverilog is correct and that is invalid verilog.  You're declaring the same signal first as an `output` and then as a `reg`.  Inputs/outputs can take a type so what you want is `output reg`, and then if you need to assign an initial value use an always block.

ericfont, 2025-02-26:
I shared the changes when I wrote "(like https://github.com/YosysHQ/nextpnr/commit/e5b44026057f86b13499ec23e083e5a1c01ee4c3)"...here is a picture of that diff:

[image]

ericfont, 2025-02-26:
For determining what pins to use, I downloaded https://www.latticesemi.com/view_document?document_id=49391 and looked at the "(SG32) 32-Pin QFN Ball Number" column, and for the LEDs picked PIO or DPIO pins starting from the bottom and going up (i.e. Ball Numbers 31, 32, 27, 26, 23 for leds 1 to 5), and for the clki picked the a GBIN pin (IOL_5A_GBIN6 aka Ball Number "8"), and put those Ball Numbers into blinky.pcf

KrystalDelusion, 2025-02-26:
Oh, sorry I didn't realise that the linked commit was the code you were using.  It looks like 68044cc fixed output wires (and potentially should have closed #328), but it doesn't address output registers.  Looks like icebox_vlog.py is the source of the issue though.

ericfont, 2025-02-26:
OK, well I guess the Yosys team has enough info from that diff I provided to reproduce the issue, so I'll let your team figure out how to fix it.

For the record I'm using the lastest [nextpnr-git	0.7.r180.g0c458f1-1 autobuilt by Arch Linux Chaotic AUR](https://aur.chaotic.cx/packages?search=nextpnr) and the [latest icestrom-git r844.7fbf8c0-1](https://aur.chaotic.cx/packages?search=icestorm), which should already have https://github.com/YosysHQ/icestorm/commit/68044cc4dac829729ccd0ee88d0780525b515746

I also have an iCE40-HX1K-VQ100 chip (on a iCE40HX1K-EVB which uses [this schematic](https://github.com/OLIMEX/iCE40HX1K-EVB/blob/master/iCE40HX1K-EVB_Rev_B.pdf)) and just now tested reassigning blinky pins according to the blinky column in this spreadsheet [iCE40PinoutHX1K.ods](https://github.com/user-attachments/files/18999014/iCE40PinoutHX1K.ods) and according to this diff https://github.com/ericfont/nextpnr/commit/5c374a7e06c1b6586db323567c3e8760b4cdf90b, but unfortunately, I get the same type of error during hte iverilog step:

`blinky_chip.v:550: error: 'io_9_0_1' has already been declared in this scope.
blinky_chip.v:3:      : It was declared here as a net.`

If there are any workarounds for when I wait for a fix to be added to the git (other than trying the proprietary iCE tools or ordering an iCEStick), or any ways I can help test, let me know.

KrystalDelusion, 2025-02-27:
I don't think the generated vlog is actually needed for programming the board, so if the problem is just in the generated vlog, you should be able to call `iceprog blinky.bin` just fine.

ericfont, 2025-02-27:
Yup...got blinky running on the Olimex iCE40HX1K-EVB!

Note for any random internet readers, that Olimex board doesn't have a flash programmer built in, but I programed it through a small cheap CJMCU FT232H USB-to-SPI bridge connected to the Olimex like https://gist.github.com/ericfont/1ed01a36e03a865dd15e567edb4c6c47 and while I couldn't quite get iceprog working I was able to use `truncate -s 2M blinky.bin` to first expand out the image size to the Olimex board's flash size and then run `flashrom -p ft2232_spi:type=232H -w blinky.bin`.